MartinL1/BMP388_DEV #1: Uninitialized structs lead to inconsistent behavior

davidjade, 2020-03-18:
Several of the structs in the BMP388_DEV class are not zero initialized and this can lead to inconsistent behavior. The unions seems especially prone to this since not all fields get set. The "config" struct was especially prone to this since it has an unnamed field. 

I only found this while trying to get it running in a native ESP-IDF project using the Arduino component layer for compatibility. In a test program I was putting the BMP388_DEV object on the stack and I would then get strange behavior. Eventually I traced the I2C with a logic analyze and could see that the initial setup config was scrambled sometimes, which I then hunted down to uninitialized structs data.

MartinL1, 2020-03-19:
Hello David,

Thank you for investigating and flagging up this bug. My apologies, I hope it didn't take up too much of your time to track down the issue?

I've now released version 1.0.1 with amended code that initially zeros the registers' structures/unions. I've also added code to allow custom I2C clock/data pins with the ESP8266 and a new setSeaLevelPressure() function that allows manual selection of the current sea level pressure.

davidjade, 2020-03-19:
Happy to help out. As I mentioned I was looking at this code and trying to make it work in non-Arduino scenarios (ESP-IDF projects targeting the ESP32 directly) so it wasn't immediately clear where the issue was. 

I've since taken a fork of this and re-mixed it into something that can be used in both ESP-IDF projects as an ESP-IDF Component as well as still being compatible with Arduino (although I have not tested that yet). It is a pretty substantial change though but one which maybe gives a bit more flexibility. I hadn't really planned on doing much with it as this is just a learning project for me, but if you'd like to see what I've done let me know and I'll clean it up. 

The main (big) thing I changed was making Device a separate base class that abstracts the concept of an I2C or SPI interface (i.e. derived I2CDevice and SPIDevice classes), which the BMP388 class then uses (via a passed-in reference, i.e. not inherited) to implement the sensor specific layer over. In theory one could just write a new Devices class for another platform and the BMP388 class would just work.

My goal was two-fold: to abstract Device into a reusable object, separating the concerns of the sensor class and secondly to provide a way to abstract away the Arduino TwoWire and SPIClass so that sensor devices could be built using any I2C/SPI APIs. 

This is a somewhat big change so your code ended up being more of a jumping off point. It also requires another thing I put together, which was a simple Arduino compatibility layer (taken from Espressif source) so that I could reuse the various Arduino sensor libraries on github in non-Arduino projects, which was my ultimate goal in starting down this path as I prefer to build things using the ESP-IDF framework and tools.

It's a work in progress but up and working so let me know if you'd like to see it.

closed 2020-08-25 by MartinL1