MartinL1/BMP388_DEV #2: Error in README.MD

glyndavidson, 2020-06-11:
See
```
**Device Configuration**
After initialisation it is possible to change the BMP388 configuration with the following functions:
bmp388.setPresOversamping(OVERSAMPING_X4);
```
setPresOversamping should be setPresOversampling
setTempOversamping should be setTempOversampling

glyndavidson, 2020-06-11:
Also TIME_STANDBY_05MS should be TIME_STANDBY_5MS

glyndavidson, 2020-06-11:
```
**Code Implementation**
if (bmp388.getMeasurements(temperature, pressure, altitude))    // Check if the measurement is complete
  {
    Serial.print(temperature);                    // Display the results    
    Serial.print(F("*C   "));
    Serial.print(pressure);    
    Serial.print(F("hPa   "));
    Serial.print(altitude);
    Serial.println(F(m"));
  }
```

Serial.println(F(m")); should be Serial.println(F("m"));

MartinL1, 2020-09-08:
@glyndavidson My apologies for the delay getting back.

I've now made the corrections to the README.MD file.

Thank you, much appreciated.

closed 2020-09-08 by MartinL1