MartinL1/BMP388_DEV #4: Not an issue - but might help with timing

glyndavidson, 2020-06-21:
In my application I need accurate timing in millisecs of each dataread, however, I'm also using sleep modes on the Arduino so can't rely on the millis() function.

To this end, I've been looking at the BMP388's internal timer.

At first glance, the .getSensorTime(); returns some odd values.

Here's the function:

`// Get the sensor time
uint32_t BMP388_DEV::getSensorTime(){
	uint32_t sensorTime;
	readBytes(BMP388_SENSORTIME_0, (uint8_t*)&sensorTime, sizeof(sensorTime - 1));
	return sensorTime & 0x00FFFFFF;
}`

However, looking into this, it seems pretty accurate.

This table shows the typical time between each read depending on what TIME_STANDBY you're using:
**Standby Time**               **Time since last read using .getSensorTime();**
TIME_STANDBY_10MS                256
TIME_STANDBY_20MS                512
TIME_STANDBY_40MS              1024
TIME_STANDBY_80MS              2048
TIME_STANDBY_160MS            4096
TIME_STANDBY_320MS            8192

You can convert the time reported by the getSensorTime() function into millisecs by dividing it by 25.6   (512/25.6=20, 1024/25.6=40), but I think this is probably the wrong way to go about it.

It seems to me that this has its roots somewhere in binary. Whilst I recognise that this probably the case, I don't have the expertise to work it out properly.
Might it have something to do with this line of code?
`return sensorTime & 0x00FFFFFF;`

glyndavidson, 2020-06-22:
Further to the above...
Simply dividing the time received by 25.6 actually gives you quite an accurate time reading, however, the sensor clock resets to 0 approximately every 10.9 minutes.  Presumably this is because the registers become full.

I made a sketch which is basically just:

`
void loop () {
  Serial.println(Sensor.getSensorTime());
}
`

The max value I got from this is 16777215, which in binary is 111111111111111111111111, so it looks like the time is stored in a 3 byte register which, as I say, fills up approximately every 10.9 minutes.
16777215 / 25.6 = 655360 millis
655360 / 1000 655.36 seconds
655.36 / 60 = 10.92267 minutes

Also of note is that when the register fills up, it sends an interrupt on the INT pin.

I haven't tested it yet, but I think you could probably use it to record accurate time using something like this:

`
unsigned long currentSensorTime, previousSensorTime;

void loop{
   // get the current time
    currentSensorTime = Sensor.getSensorTime() / 25.6;

   // check if the BMP388's time register has overflowed
   // and if it has, add on the previously recorded time
    if (currentSensorTime < previousSensorTime){
        currentSensorTime += previousSensorTime;
    }
   
    // record the current time for comparison in the next loop
    previousSensorTime = currentSensorTime ;
}
`

This isn't perfect because dividing by 25.6 into an unsigned long you are going to lose some accuracy. You could of course do some math and find a workaround for this too.

Anyway, I'm going to keep playing with this, but for the moment, I have found that if I just add the TIME_DELAY every time the data is ready, I get more accurate time keeping than I do using the Arduino's millis() function...

For example:

```
volatile boolean dataReady = false;
unsigned long time = 0;

void sensorInterruptServiceRoutine() {
  dataReady = true;
}

void setup() {
  attachInterrupt(0, sensorInterruptServiceRoutine, FALLING);
  Sensor.enableInterrupt();
  Sensor.usingInterrupt(0);
  Sensor.setTimeStandby(TIME_STANDBY_80MS);
  Sensor.startNormalConversion();
}

void loop{
   if(dataReady){
      time += 80;   // TIME_STANDBY_80MS
   }
}
```

glyndavidson, 2020-06-23:
Further to the above, here's a sketch which demonstrates how to keep accurate time using the BMP388's internal clock.
I've tested this over a 12 hour period and found it to be accurate. Over a longer period you may need to adjust.

```
/*   Sketch which demonstrates how to use the BMP388 internal clock for "accurate" time keeping
 *    Useful for when you are using sleep modes and want to keep track of time -
 *    without having to use a Real Time Clock
 */
#include <BMP388_DEV.h>

// Using SPI mode
#define BMP_CS_PIN 2
BMP388_DEV  Sensor(BMP_CS_PIN );

// Duration (in seconds) to compile and upload your sketch - Adjust for your particular setup 
#define SECS_TO_COMPILE_AND_UPLOAD 11

// Set the current clock according to your programming environment
const char compilerTime[] = __TIME__;
const unsigned long hoursToMils = atoi (compilerTime + 0) * 60UL * 60UL * 1000UL;
const unsigned long minsToMils = atoi (compilerTime + 3) * 60UL * 1000UL;
const unsigned long secsToMils = atoi (compilerTime + 6) * 1000UL;
const unsigned long timeStart = hoursToMils + minsToMils + secsToMils + (SECS_TO_COMPILE_AND_UPLOAD * 1000UL);

unsigned long currentTime, previousTime, overflowTime;

float temperature, pressure;

void setup(){
   // Start the Serial Monitor
   Serial.begin(115200);

   // Start the Sensor in normal mode
   // and with 1280ms intervals (The closest the BMP388 has to one second)
   if (Sensor.begin()) {
     Sensor.setTimeStandby(TIME_STANDBY_1280MS);
     Sensor.startNormalConversion();
  }
}

void loop () {
   // if Sensor data is ready (every 1280ms)
   if(Sensor.getTempPres(temperature, pressure)){

      // The BMP388 stores time in binary format so...
      // Convert the time reported by the sensor into millisecs
      currentTime = Sensor.getSensorTime() / 25.6;
    
      // The BMP388 time registers fills up and restarts approx every 10.9 minutes
      // So every time it overflows, we need to increment our time keeping
      if(currentTime < previousTime){
         // I've tested this figure over a 12 hour period and found it to be accurate.
         // if you test it for a longer period, and find that your timer is out then adjust this value.
         // increase the value if you lose time, decrease it if you gain time. 
         overflowTime += 655844;
         Serial.println("# # # - OVERFLOW - # # #");
      }

      // Keep track of time
      previousTime = currentTime;
      currentTime += overflowTime;    
      currentTime += timeStart;

      // Convert millis into hours, mins and secs
      currentTime /= 1000;
      hours = currentTime / 3600;
      int secsRemaining = currentTime % 3600;
      mins = secsRemaining / 60;
      secs = secsRemaining % 60;

       // Print time in hh:mm:ss format
      Serial.print(hours);
      Serial.print(F(":"));
      if (mins < 10) Serial.print(F("0"));
      Serial.print(mins);
      Serial.print(":");
      if (secs < 10)  Serial.print(F("0"));
      Serial.println(secs);
   }
}
```

MartinL1, 2020-09-08:
@glyndavidson That's an ingenious use of the BMP388's timer.

The overflow of 10.92 minutes makes sense, since the barometer's maximum delay is 655360ms.

Following you line of though, dividing 655360ms by the timer's 24-bit count register at 2^24, gives a timer tick of 39.0625us. The inverse gives 1/39.0625us=25600 ticks per second, dividing further by 1000 gives 25.6 ticks per millisecond. Therefore, dividing the barometer's count register by 25.6 gives the number of elapsed milliseconds or millis.

Thank you Glyn for explanation, that's really interesting way keeping track of time, while simultaneously placing a MCU without a RTC in sleep mode.

closed 2020-09-08 by MartinL1