MartinL1/BMP388_DEV #10: Field Definitions

myval, 2021-01-17:
Hi,
I have used your BMP280 library in the past, I just got BMP388 and I thought I would use the same library maker but Iam either missing something or there is couple of errors in the library. The main problem I am having at the moment is that if I set this
`bmp388.setTimeStandby(TIME_STANDBY_40MS);`
it will never set the interrupt in normal mode Basically dont get any measurements if I have any overampling on.

Then those Field defs:
`OVERSAMPLING_SKIP 		 		= 0x00,  
	OVERSAMPLING_X1   		 	= 0x01,
	OVERSAMPLING_X2   		 	= 0x02,
	OVERSAMPLING_X4  		   	= 0x03,
	OVERSAMPLING_X8    		 	= 0x04,
	OVERSAMPLING_X16   	 	 	= 0x05,
	OVERSAMPLING_X32   	 	 	= 0x06`

should be like this:
`	OVERSAMPLING_SKIP 		 	= 0x00,  
	OVERSAMPLING_X2   		 	= 0x01,
	OVERSAMPLING_X4  		   	= 0x02,
	OVERSAMPLING_X8    		 	= 0x03,
	OVERSAMPLING_X16   	 	 	= 0x04,
	OVERSAMPLING_X32   	 	 	= 0x05`

and just a cosmetic bit really to line it up with the datasheet
`	IIR_FILTER_OFF  			= 0x00,
	IIR_FILTER_1    			 	= 0x01,
	IIR_FILTER_3           			= 0x02,
	IIR_FILTER_7           			= 0x03,
	IIR_FILTER_15          			= 0x04,
	IIR_FILTER_31          			= 0x05,
	IIR_FILTER_63          			= 0x06,
	IIR_FILTER_127         		= 0x07`

MartinL1, 2021-01-18:
Hi myval,
Thanks for the correction to the oversampling definitions, I mistakenly copied it over from the BMP280_DEV library. I'll make the correction then update the library.

In respect of the filter co-efficient definition settings, it depends where you look in the BMP388 datasheet. In tables 7 and 9, the coefficient values are listed as 2^n for values >0: off, 2, 4, 8, 16 & 32. In other sections such as the resgister description, it states that they're 2^n - 1: 0, 1, 3, 7, 15 & 31. I just happened to choose the former.

Regarding the timing, I believe it's because you're configuring the device oversampling settings for a measurement time that's longer than the barometer's sample time. See BMP388 datasheet table 3.9.1 Measurement Time.

I've set up and successfully tested my Arduino Uno with the BMP388. I ran this code below that configures the barometer for I2C operation, Normal Mode, External Interrupt with no temperature oversampling, x8 pressure oversampling (with the oversampling definition correction). I also successfully tested it with the temperature oversampling set to x2:

```
//////////////////////////////////////////////////////////////////////////////////////////
// BMP388_DEV - I2C Communications, Default Configuration, Normal Conversion, Interrupt
//////////////////////////////////////////////////////////////////////////////////////////

#include <BMP388_DEV.h>                           // Include the BMP388_DEV.h library

volatile boolean dataReady = false;               // Define the data ready flag
float temperature, pressure, altitude;            // Declare the measurement variables

BMP388_DEV bmp388;                                // Instantiate (create) a BMP388_DEV object and set-up for I2C operation (address 0x77)

void setup() 
{
  Serial.begin(115200);                           // Initialise the serial port
  bmp388.begin();                                 // Default initialisation, place the BMP388 into SLEEP_MODE 
  bmp388.enableInterrupt();                       // Enable the BMP388's interrupt (INT) pin
  attachInterrupt(digitalPinToInterrupt(2), interruptHandler, RISING);   // Set interrupt to call interruptHandler function on D2
  bmp388.setTempOversampling(OVERSAMPLING_SKIP);  // Set for no temperature oversampling
  bmp388.setPresOversampling(OVERSAMPLING_X8);    // Set for x8 pressure sampling
  bmp388.setTimeStandby(TIME_STANDBY_40MS);         // Set the standby time to 40ms
  bmp388.startNormalConversion();                     // Start BMP388 continuous conversion in NORMAL_MODE 
}

void loop() 
{
  if (dataReady)                                      // Check if data is ready
  {  
    bmp388.getMeasurements(temperature, pressure, altitude);      // Read the measurements
    Serial.print(temperature);                    // Display the results    
    Serial.print(F("*C   "));
    Serial.print(pressure);    
    Serial.print(F("hPa   "));
    Serial.print(altitude);
    Serial.println(F("m")); 
    dataReady = false;                            // Clear the dataReady flag
  }   
}

void interruptHandler()                           // Interrupt handler function
{
  dataReady = true;                               // Set the dataReady flag
}
```
The oversampling settings (no temperature, x8 pressure) keeps the device measurment time to (typically) 19.2ms, as specified in table 3.9.1 Measurement Time table in the datasheet.  This is well within the 40ms sample time specified.

myval, 2021-01-20:
Hi,
with the coefficients are you sure you run latest manual? They have changed this in manual V1.1
I have the same settings as in your example and it does not work when I set it  to 80ms it works as expected.  Wondering if Bosch have not screwed up anything in the latest batch of sensors. As soon as I have a minute I will try to play with it again.

MartinL1, 2021-01-20:
> with the coefficients are you sure you run latest manual? They have changed this in manual V1.1

Thanks for the update. Uploading their latest datasheet, I can see that they've changed it.

Although I need to correct the oversampling definition error that you raised, I'm reluctant to change the IIR_Filter definitions, as any library update will generate syntax errors for users whose code previously worked.

> I have the same settings as in your example and it does not work when I set it to 80ms it works as expected. Wondering if Bosch have not screwed up anything in the latest batch of sensors. As soon as I have a minute I will try to play with it again.

Have you made the necessary change to the oversampling definitions in the "BMP388_DEV.h" file?:

```
enum Oversampling {
	OVERSAMPLING_SKIP 		 = 0x00,     			// Oversampling bit fields in the control and measurement register
	OVERSAMPLING_X2   		 = 0x01,
	OVERSAMPLING_X4  		 = 0x02,
	OVERSAMPLING_X8    		 = 0x03,
	OVERSAMPLING_X16   	 	 = 0x04,
	OVERSAMPLING_X32   	 	 = 0x05
};
```

Might I ask which microcontroller board you're using and whether your using I2C or SPI comms? This will give me the opportunity to see if I can replicate the issues you're encountering? 

I'm currently testing using an Arduino Uno and an Adafruit BMP388 breakout board.

myval, 2021-01-20:
I did change it all in my "BMP388_DEV.h" just so I have it in line with my datasheet. I guess it is fine if it stays the way it is. If someone new gets confused will have trace here anyway.
I have my own board with atmega328PB and two BMP388s and I actually tried it just now it worked so beats me what happened. I have tried it like 10 times few days ago but now it all works so I feel like an idiot. All I did was I coppied your settings from message above which worked then I went back to my settings with 40ms and it works too I dont know. Some kind of gremlin which got chased away with your settings.

MartinL1, 2021-01-21:
Hi myval,
Glad to hear that everything's now working. It's like that with electronics sometimes...no obvious rhyme or reason.

Many thanks for the correction, I'll update the library shortly.

myval, 2021-01-21:
certainly it is like that all the time, one step forwared and fast run back.
Thanks for the great library.
Josef

myval, 2021-01-22:
Mate,
one more thing. I was wondering why is the library loading spi.h as well as wire.h if I run it on I2c?

MartinL1, 2021-01-23:
>  I was wondering why is the library loading spi.h as well as wire.h if I run it on I2c?

Unfortunately, it's going to be a somewhat complex and lengthy answer a simple, but interesting question, that certainly got me thinking.

This issue here is that the BMP388_DEV library is working within the Arduino framework. As the library supports both I2C and SPI, it needs access to the functions of both. This means including both <Wire.h> and <SPI.h>. However, the act of simply including these libraries, causes the Arduino code to automatically instantiate (create) both a Wire and SPI class object in the microcontroller's memory. Arduino did this, to save their users from having to manually call the respective Wire and SPI class constructors at the beginning of their sketch. This though, is somewhat inefficient from a memory perspective, if one of them isn't subsequently used at run time, as is the case here. The problem is that the BMP388_DEV library doesn't know which of them will be called upon until run time, (rather than at the time of code compilation), so it has to call on both.

It's possible to overcome this issue, either by creating two separate BMP388 libraries, one for I2C the other for SPI, or alternatively have a precompiler switch out the unwanted code at compile time, which turns out to be quite an ugly solution.

For most microcontrollers having the slight inefficiency of creating a memory footprint for both the I2C and SPI objects isn't a problem, however I appreciate that this can be an issue if you've got an AVR microcontroller with about 32KB of flash and 2KB of RAMs, where memory starts to become a precious resource.

One way to achieve this, would be to comment out all the lines that reference SPI in the "Device.h" and "Device.cpp" files and the SPI class constructors in the "BMP388_DEV.h" and "BMP388_DEV.cpp" files.

myval, 2021-01-23:
ok, thanks for the answer. I am getting to the stage that I might run out of memory, that is why I have started to look at unnecessary libraries being loaded and saw the SPI which I dont use. 
I actually already started to look at slimming down your library and get rid of the SPI sections, but I wanted to see if there is not an easier path.  
thank you for your help
Josef

MartinL1, 2021-01-23:
Hi Josef,
Just updated the oversampling definitions. A new update should be available in the Arduino IDE Library Manager shortly.

closed 2021-01-28 by MartinL1