MartinL1/BMP388_DEV #11: why using int_status instead of status?

Alain2019, 2021-04-04:
Hi

I have some strange errors while using you're library witg a BMP390L.  Sometimes I get a error result.  At the start I kept going wrong, but a extra begin before every forced reading solved that.

Looking at the code I see that you use INT_STATUS and not STATUS (and ERR) for datready, is there a reason for?

MartinL1, 2021-04-05:
Hi Alain2019,

> I have some strange errors while using you're library witg a BMP390L. Sometimes I get a error result. At the start I kept going wrong, but a extra begin before every forced reading solved that.

Are you using the BMP390L in forced mode for non-blocking operation? If this is the case, how long does a single pass of your loop() function take?

> Looking at the code I see that you use INT_STATUS and not STATUS (and ERR) for datready, is there a reason for?

Because the _drdy_ bit in the INT_STATUS register ties in more closely with the BMP388's interrupt operation and the fact that other than being shown in the timing diagrams, the STATUS register isn't discussed anywhere in the datasheet.

Alain2019, 2021-04-05:
Hi forced mode for non-blocking (but in the same loop), I"ll add more code later.

The code inside one loop:

delay(3);  // and binking led
rtcY2kEpoch = rtc.getY2kEpoch();
    delay(9);  // 2 of 3 msec would be enough according to the datasheet
    bmp388.begin(BMP388_I2C_ALT_ADDR);                                 //
Default initialisation, place the BMP388 into SLEEP_MODE
    bmp388.setPresOversampling(OVERSAMPLING_X8);  
    bmp388.setTempOversampling(OVERSAMPLING_X2);  
    bmp388.setIIRFilter(IIR_FILTER_OFF);

    bmp388.startForcedConversion();                 // Start BMP388 forced conversion (if we're in SLEEP_MODE)
    delay(22);  // 8x pressure en 2x temp max 22msec  Will do some other things later in this time.
    milliseconden = 22;
    while (((result_TempPres = bmp388.getTempPres(temperature, pressure)) == 0) && (milliseconden < 100))
    {
      delay(1);
      ++milliseconden;
    }
// on error I get milliseonden == 100
deepsleep(60seconds); // samd21 goes in a deep sleep.

errors happen less than 1/4.
2 error candidates :
- Border case problems I²C pullup, I already added 2 extra pull-ups.  I could make them still lower.
- Is the INT_STATUS cleared to early, or other problem.

MartinL1, 2021-04-06:
Hi Alain2019,

I've tested the following code on my SAMD21 board and BMP388 breakout using the I2C communications, but I'm unable to reproduce the issue: 

```
///////////////////////////////////////////////////////////////////////////////
// BMP388_DEV - I2C Communications, Default Configuration, Forced Conversion
///////////////////////////////////////////////////////////////////////////////

#include <BMP388_DEV.h>                           // Include the BMP388_DEV.h library

float temperature, pressure, altitude;            // Create the temperature, pressure and altitude variables
BMP388_DEV bmp388;                                // Instantiate (create) a BMP388_DEV object and set-up for I2C operation (address 0x77)

void setup() 
{
  SerialUSB.begin(115200);                           // Initialise the serial port
  while(!SerialUSB);
  bmp388.begin();                                 // Default initialisation, place the BMP388 into SLEEP_MODE 
  bmp388.setPresOversampling(OVERSAMPLING_X8);
  bmp388.setTempOversampling(OVERSAMPLING_X2);
  bmp388.setIIRFilter(IIR_FILTER_OFF);
  //bmp388.setIntLatchConfig(LATCHED);
}

void loop() 
{
  bmp388.startForcedConversion();                 // Start BMP388 forced conversion (if we're in SLEEP_MODE)
  delay(22);  // 8x pressure en 2x temp max 22msec  Will do some other things later in this time.
  uint32_t milliseconden = 22;
  while (((bmp388.getTempPres(temperature, pressure)) == 0) && (milliseconden < 100))
  {
    delay(1);
    ++milliseconden;
  }
  SerialUSB.print(millis());
  SerialUSB.print(F("ms   "));
  SerialUSB.print(milliseconden);
  SerialUSB.print(F("ms   "));
  SerialUSB.print(temperature);                    // Display the results    
  SerialUSB.print(F("*C   "));
  SerialUSB.print(pressure);    
  SerialUSB.println(F("hPa   "));
  delay(1000);                                    // Wait for 1 second
}
```
I'm getting a conversion time of around 21ms. If the milliseconden delay is taken below this, it remains at 21ms, as expected. Any value above this simply takes the specified time in milliseconds, for example if I specify a 40ms delay, milliseconden returns 40ms. Again that's what I'd expect, since in this case the bmp388.getTempPres() function returns immediately with the pressure and temperature results.

Alain2019, 2021-04-06:
Thanks for testing.

I also get the error not that often.  I'm running now so that I get the error and status register when there are problems and every 20 minutes when ok.

Status>>4 is 7 when pressure and temp are ready, 1 10ms after the forced command before there ready.  Error is 0.

Alain2019, 2021-04-07:
After testing the failed readings seems to be always have also an error code.  But the error code is strange and is using the reserved bits and sometimes even only reserved bits aka 96 = 1100000  ....

Maybe a problem after the deepsleep.  I will add a retry and see if that does help.

MartinL1, 2021-04-07:
Hi Alain2019,

I've released a new version of the BMP388_DEV (version 1.0.6) that allows the STATUS and ERROR registers to be read with the functions getErrorReg() and getStatusReg(). This should become available as an update in the Arduino IDE's Library Manager shortly.

I've repeated the same code, but also outputting the ERROR and STATUS registers:
```
///////////////////////////////////////////////////////////////////////////////
// BMP388_DEV - I2C Communications, Default Configuration, Forced Conversion
///////////////////////////////////////////////////////////////////////////////

#include <BMP388_DEV.h>                           // Include the BMP388_DEV.h library

float temperature, pressure, altitude;            // Create the temperature, pressure and altitude variables
BMP388_DEV bmp388;                                // Instantiate (create) a BMP388_DEV object and set-up for I2C operation (address 0x77)

void setup() 
{
  SerialUSB.begin(115200);                           // Initialise the serial port
  while(!SerialUSB);
  bmp388.begin();                                 // Default initialisation, place the BMP388 into SLEEP_MODE 
  bmp388.setPresOversampling(OVERSAMPLING_X8);
  bmp388.setTempOversampling(OVERSAMPLING_X2);
  bmp388.setIIRFilter(IIR_FILTER_OFF);
  //bmp388.setIntLatchConfig(LATCHED);
}

void loop() 
{
  bmp388.startForcedConversion();                 // Start BMP388 forced conversion (if we're in SLEEP_MODE)
  delay(40);  // 8x pressure en 2x temp max 22msec  Will do some other things later in this time.
  uint32_t milliseconden = 40;
  while (((bmp388.getTempPres(temperature, pressure)) == 0) && (milliseconden < 100))
  {
    delay(1);
    ++milliseconden;
  }
  SerialUSB.print(millis());
  SerialUSB.print(F("ms   "));
  SerialUSB.print(milliseconden);
  SerialUSB.print(F("ms   "));
  SerialUSB.print(bmp388.getErrorReg(), HEX);
  SerialUSB.print(F("   "));
  SerialUSB.print(bmp388.getStatusReg(), HEX);
  SerialUSB.print(F("   "));
  SerialUSB.print(temperature);                    // Display the results    
  SerialUSB.print(F("*C   "));
  SerialUSB.print(pressure);    
  SerialUSB.println(F("hPa   "));
  delay(1000);                                    // Wait for 1 second
}
```
I ran this for over half an hour without any issues. The ERROR register returning 0x00 and the STATUS returning 0x10, indicating that the barometer was ready to receive a new command.

Alain2019, 2021-04-10:
Thanks

i'm waiting to get some extra components.  It could be a component problem.

BTW I used a deepsleep, which puts the samd21 itself in a powerdown state.  The rtc does restart it.

MartinL1, 2021-04-11:
Hi Alain2019,

I tried running the same code, but using the ArduinoLowPower library and a Serial1 with a Serial-To-USB converter, to keep the serial port alive after deep sleep. The ArduinoLowerPower library uses the RTC to keep time while the CPU is asleep. The _LowerPower.deepSleep(1000)_ function replacing the _delay(1000)_:

```
///////////////////////////////////////////////////////////////////////////////
// BMP388_DEV - I2C Communications, Default Configuration, Forced Conversion
///////////////////////////////////////////////////////////////////////////////

#include <BMP388_DEV.h>                           // Include the BMP388_DEV.h library
#include <ArduinoLowPower.h> 

float temperature, pressure, altitude;            // Create the temperature, pressure and altitude variables
BMP388_DEV bmp388;                                // Instantiate (create) a BMP388_DEV object and set-up for I2C operation (address 0x77)

void setup() 
{
  Serial1.begin(115200);                           // Initialise the serial port
  //while(!Serial1);
  bmp388.begin();                                 // Default initialisation, place the BMP388 into SLEEP_MODE 
  bmp388.setPresOversampling(OVERSAMPLING_X8);
  bmp388.setTempOversampling(OVERSAMPLING_X2);
  bmp388.setIIRFilter(IIR_FILTER_OFF);
  //bmp388.setIntLatchConfig(LATCHED);
}

void loop() 
{
  bmp388.startForcedConversion();                 // Start BMP388 forced conversion (if we're in SLEEP_MODE)
  delay(40);  // 8x pressure en 2x temp max 22msec  Will do some other things later in this time.
  uint32_t milliseconden = 40;
  while (((bmp388.getTempPres(temperature, pressure)) == 0) && (milliseconden < 100))
  {
    delay(1);
    ++milliseconden;
  }
  //Serial1.print(millis());
  //Serial1.print(F("ms   "));
  Serial1.print(milliseconden);
  Serial1.print(F("ms   "));
  Serial1.print(bmp388.getErrorReg(), HEX);
  Serial1.print(F("   "));
  Serial1.print(bmp388.getStatusReg(), HEX);
  Serial1.print(F("   "));
  Serial1.print(temperature);                    // Display the results    
  Serial1.print(F("*C   "));
  Serial1.print(pressure);    
  Serial1.println(F("hPa   "));
  //delay(1000);                                    // Wait for 1 second
  LowPower.deepSleep(1000);
}
```
Again the board ran without any issues for over an hour.

closed 2021-04-29 by MartinL1