MartinL1/BMP388_DEV #12: Compilation problem with Arduino DUE

claudius310, 2021-04-23:
hello,
Cannot use this code with Arduino DUE in I2C.
I have a message "Compilation error for the Arduino DUE board"
Do you have a solution?
Regards

MartinL1, 2021-04-29:
@claudius310 My apologies for the delay in getting back to you.

Thank you for flagging up this issue. The reason for the compilation error on the Due was that unlike other Arduinos, it doesn't implement the SPI functions: usingInterrupt() and notUsingInterrupt() functions.

I've now fixed and tested it. The updated BMP388_DEV library is available as version 1.0.7.

claudius310, 2021-05-01:
Hello,
Thank you for this modification which works well in I2C and SPI With Arduino DUE
But in SPI it is necessary to connect
SCK BMP388: Arduino DUE SCK, D76
SDI BMP388: Arduino DUE MOSI, D75
SDO BMP388: Arduino DUE MISO D74
CS BMP388: Arduino DUE D10
Best regards 
thank you very much