MartinL1/BMP388_DEV #13: Unstable communication with 6 sensors on the same SPI bus

Emanuel-Pytlik, 2021-11-17:
Hello,

This software works out wonderful with only one sensor for me. However, when i want to read out 6 sensors at the same time the communication either is unstable or does not work at all. According to my observations, the problem encounters during the execution of the "getFIFOData" function. I am using the Bosch BMP384 Sensors, the Arduino IDE and a Teensy 3.2. I attached my code here:
[Bracelet.txt](https://github.com/MartinL1/BMP388_DEV/files/7553946/Bracelet.txt)

Thanks for Your answer in advance!

glyndavidson, 2021-11-17:
My guess would be conflicting i2c addresses? 
Consider using an i2c multiplexer.

MartinL1, 2021-11-17:
Hi @Emanuel-Pytlik 

Wow, six barometers! I've done some testing, however I've only got one, single BMP388 sensor to play with.

Initiating temperature/pressure oversampling and additional filtering adds delays to each measurement. In normal mode using fast standby times together with oversampling and filtering, it's possible to instruct the barometer to sample at a rates that it's incapable of meeting. Whenever this happens, the getFIFOData() function is unable to return any meaningful data.

I'd suggest starting by turning off the temperature/pressure oversampling and filtering, this should at least allow your group of barometers to sample at these higher rates and see if this works?

Emanuel-Pytlik, 2021-11-18:
> My guess would be conflicting i2c addresses? Consider using an i2c multiplexer.

I am using an SPI bus.

Emanuel-Pytlik, 2021-11-18:
Thank you @MartinL1 for your quick reply!
Turning off temperature and pressure oversampling and the filtering also doesn't work.
According to this table [image] i should be able to use a standby time of 10ms with my settings but i am only using a standby time of 20ms. I somehow have the feeling that sensors go back to sleep if the FIFO is not read for a while. If i for instance use only one sensor with a really short standby time and a long delay between the iterations of the loop, it also does not work. This code for example does not work:
[Test.txt](https://github.com/MartinL1/BMP388_DEV/files/7561173/Test.txt)
However, if i change the LOOP_DELAY to 0ms, it works just fine. Can you verify my guess?

MartinL1, 2021-11-18:
Hi @Emanuel-Pytlik

I trawled back through the library code to see what's going on, since it has been a while since I looked at it.

The reason why the code works without a _delay()_ function and doesn't with, is because the _delay()_ prevents the FIFO from being serviced. 

When initialising the FIFO, the _setFIFONoOfMeasurements()_ function sets the FIFO's "watermark" to the number of expected bytes required to store the specified number of measurements. Whenever the last measurement is taken it triggers the FIFO watermark interrupt flag, indicating that the FIFO is ready to be read. The _loop()_ function needs to keep up with the FIFO, otherwise it will quickly overflow.

Also, access to the FIFO is limited in that there's no direct access to any of its memory locations and each group of measurements has to be read with a single burst read.

Once way to ensure that each FIFO is serviced, is to block and wait for each barometer's measurements using a _while()_ statement rather than using _if()_:
```
while (!bmp388_1.getFIFOData(temperature, pressure_1, altitude, sensorTime));   // Wait for sensor 1's FIFO to be ready
```