MartinL1/BMP388_DEV #14: BMP280 and BMP388 do not work at the same time

msfujino, 2022-01-16:
In order to compare the performance of BMP280 and BMP388, I need to run them at the same time, and I have two devices connected to the I2C bus.
I have included <BMP280_DEV.h> and <BMP388_DEV.h>, but as I feared, the compilation does not go through. When I read the library, I found many similar definitions, and I imagine this is the cause.
If there is any easy way to solve this problem, please let me know.

MartinL1, 2022-01-16:
Hi @msfujino 

Thanks for the message. I'll look in it.

msfujino, 2022-01-16:
For reference, I have attached the file I am trying to compile. I have tried Arduino_nano and Seeed_XIAO devices, and both of them give me many error messages like below.
error: multipule definition of 'enum SPIPort'
Also, if I change the order in which the two include files are written, I get different error messages.

Thank you very much for your help
[BMP388_BMP280_I2C_Normal.txt](https://github.com/MartinL1/BMP388_DEV/files/7878102/BMP388_BMP280_I2C_Normal.txt)
.