MartinL1/BMP388_DEV #15: implement a convenient way to let the user provide a pointer to an I2C device as an interface to the BMP

jerabaul29, 2022-03-15:
First, thank you very much for a very nice library.

We are looking at this library together with @jvoermans (cc:ing :) ).

It is a bit difficult to provide a custom ```Wire``` object to perform the I2C communications on general arduino boards (outside of a few STM boards for which you provide some way to set the ```scl``` and ```sda```). For example, on our board, we have a number of wire ports, and we would like to use one specific for the BMP.

The way we are going to do it "on our side" is to hack things, replacing the calls to ```Wire``` in https://github.com/MartinL1/BMP388_DEV/blob/master/Device.cpp by calls to our own ```myWire``` object; but I think that it would be great on the longer term and for other users if it would be possible to provide a way to do this simply at some point in the future, for example, by adding some constructor that take in a ```Wire *``` or similar :) .

MartinL1, 2022-06-28:
@jerabaul29 My apologies for the delay in getting back. 

Thank you for your excellent suggestion. I've now added the functionality to specify a secondary or auxiliary I2C port.

The secondary Wire library is passed by reference in the constructor, so for example on the Arduino Due's Wire1 library:

```
BMP388_DEV bmp388(Wire1);  // Instantiate (create) a BMP388_DEV object and set-up for I2C operation (address 0x77) on Wire1
```

jerabaul29, 2022-08-03:
No worries :) . Many thanks for the implementation :) . @jvoermans , this will make it much easier to use this sensor and library in future projects, FYI :) .

closed 2022-08-03 by jerabaul29